InsanityAutomation/Marlin #207: firmware_Ender6_BLT_DW7.4.bin causes crashing into bed

JCSWeb, 2021-06-30:
When testing this firmware, the probe extends but then crashes into the bed.

BL Touch is connected via the splash / breakout board provided with the Ender 6.

JCSWeb, 2021-07-01:
So it seems using this firmware it expects the BL Touch to be wired as the Z endstop switch which the filename structure doesn't suggest. After rewiring to utilise the Z stop I managed to get the printer to home.

Mesh leveling now seems to fail though. When initially going onto the mesh process, it attempts to start, homes, then kicks back out and stops without any messages.

kathagor, 2021-07-01:
I can confirm the same problem. The Head is moving to the center position and the bltouch pin is deployed, but no reaction on hiting the pin. The pin is not retracted on hit.

But printer is reacting to the Z-Endstop. So probably the wrong pin is configured.

InsanityAutomation, 2021-07-03:
I'm putting this machine back on the bench this morning, so hopefully it can get sorted quickly :)

kathagor, 2021-07-03:
👍 Super. i think it is only the wrong pin mapped for the bltouch, because
ender6 uses a dedicated pin for the bltouch. And the printer reacts to the
z endstop like it Was the bltouch.

InsanityAutomation, 2021-07-03:
Updated hex files have been uploaded! Please let me know if any issues persist.

subhead, 2021-07-03:
> Updated hex files have been uploaded! Please let me know if any issues persist.

Where can i find these updated hex files?

JCSWeb, 2021-07-03:
> Updated hex files have been uploaded! Please let me know if any issues persist.

Perfect, Will test when I can. On holiday right now (no, I wasn't allowed to bring the printer) so will try and report back when I can 👍

subhead, 2021-07-03:
Alright nevermind. Found the branch with the hex files.

subhead, 2021-07-03:
I updated my Ender 6 with the latest  BLTouch version. Some issues i found so far:

- Auto Leveling can not toggled on. After pressing on it it still remains off.
- After pressing the  Z +/- buttons the bed did not move closer or further away from the nozzle. Instead it only shows on the screen that the new z offset is written. With that no initial bed leveling is possible.

InsanityAutomation, 2021-07-04:
> I updated my Ender 6 with the latest BLTouch version. Some issues i found so far:
>
>     * Auto Leveling can not toggled on. After pressing on it it still remains off.
>
>     * After pressing the  Z +/- [...]

I didnt see that behavior earlier today, i have it set with the z offset over 2mm off so i can see the drastic move when testing that.  Any chance you can get a serial log from that occurring? The leveling state wont turn on if the mesh isnt valid, but that should be reported in the log as well. I usually totally ignore the button for it personally, let start gcode call it on for a safety anyway.

banpot, 2021-07-17:
> I didnt see that behavior earlier today, i have it set with the z offset over 2mm off so i can see the drastic move when testing that. Any chance you can get a serial log from that occurring? The [...]

I face the same trouble, when pressing measure icon, it's look freezing my ender6 for a few minutes, cannot press any button. After minutes of waiting, the mesh measurement continue to works like a charm.

Toggleing Auto Leveling has the same problem as @subhead mentioned. but after waiting for mesh measuring for few minutes, it automatically toggled to ON.

InsanityAutomation, 2021-07-17:
@banpot Have the screen files been flashed? There should be a bed heating message after pressing measuring that indicates what its doing / waiting for. It will reject other tasks until the current task is complete.

notfixingit3, 2021-08-11:
1) The screen flashes for a second (does not show heating message and appears to freeze). After waiting (can see temp rise on octoprint) it does its thing.

2) It never shows the z offset when adjusting and it does not seem to save it, also when exiting out of leveling and going back in, it slams the nozzle into the bed or when pressing home, same issue.

3) Z offset buttons are backwards in the leveling menu but correct in the move menu

Maybe I missed the screen flashing bit? Would that cause the above issues?

InsanityAutomation, 2021-08-11:
The Z arrows were the source of some debate. Currently, the motion is consistent with the axis position being increased or decreased. Users from the older machines (Ender 5 Plus) or from CNC operations were expecting this operation. The ender 6, Creality swapped the 2 buttons from the screen side. Currently, the direction on those is by design, so not a bug though I do understand its different than how it was on this machine originally.

notfixingit3, 2021-08-11:
Ah ok, that makes sense, thank you! I think just using the stock Ender 6 since I got it and not having experience with others, my brain just logically defaults to up is up and down is down. Nots sure if it's possible, but maybe an option to invert the controls (or maybe it's not worth it)? This makes me realize now why I had to invert Z in octoprint originally