InsanityAutomation/Marlin #209: Auto leveling make pauses and shows "Bed heating" cr10s Pro V1 ( capacitive sensor )

shifubrams, 2021-07-04:
### Description
I upgraded from DW6 to DW7.4 and now the leveling is taking lot of time. 
Before each point ( 25 ) it waits the temp of the bed and shows "bed heating" but the bed temperature is stable. 
Is this a known issue or it is a wanted behaviour ? 

### Steps to Reproduce

1. install DW4.7 on cr10s Pro V1
2. run a bed leveling
3. see the pauses between each points
**Expected behavior:** shouldnt pause on each point. ( too long )
**Actual behavior:** make a pause before each point
#### Additional Information
I installed from the .hex file

[image]

CenGo86, 2021-07-05:
Same issue with TH3D EzABL on Ender 6

InsanityAutomation, 2021-07-07:
This is expected, and ensures a more accurate probe with the capacitive sensor. Bed heating will cause deviation in the trigger point. Since the intention is to probe once accurately and store to eeprom, this should only cause a delay when re-calibrating due to some change.

DerRudi79, 2021-08-02:
Hi,
Can i turn off the stop at the level?

MarcusBenjamin, 2021-08-28:
CR-10S Pro updated today, it does this with EVERY print.  This is not acceptable nor normal.  I did a PID tune on the hot end and it heats up irregularly, more deviation that before on 7.3.1 from Tiny Machines.  A few degrees isn't going to mess up anything, but the time this is taking is over 5 minutes, which used to take maybe 1 minute. 
Please fix was was done from 7.3.1 to 7.4 in this regard.  Thanks
[image]
[image]
[image]
[image]

MyStoopidStuff, 2021-09-26:
I was happy to have found this post, since I have been questioning what happened to my CR10S Pro (v1 with the stock capactive probe) after upgrading to DW7.4, since now an ABL takes 15-20m, due to waiting for the bed during probing.  During ABL the printer pauses ever 1 or 2 probe points due to the heating, which seems to demand perfection now.  I had run the bed PID tune multiple times, but it did not help, so I hope this is fixed in the next release.

To avoid the 15-20m delay in starting a print, I no longer ABL every time and use the stored mesh unless I have a very large print (commented out the G29 I was using):

G28 ;Home
;G29 ; initiate z-probing
M420 S1 Z2 ;Enable ABL using saved Mesh and Fade Height

thetailwind, 2021-11-07:
I believe this should be closed. I updated from DW7.4 to DW7.4.3 and the issue has gone away. 

Tested by community.

InsanityAutomation, 2021-11-07:
I'll make a sweep through and close stuff in a few days, trying to finish up adding Sermoon and Ender 7 support as well as a few other tidbits in the next few days!