InsanityAutomation/Marlin #211: Unable to proceed after refuelling filament

cai876, 2021-07-08:
### Description

Hi, I have tried printing a large object which requires more than one filament reel to print. After my filament ran out I got a message saying there is no filament left and click yes to change filament. I selected yes and nothing happens and it displays the message heater timeout. Previously on this printer and on my CR-200B, I select print and it heats the extruder up and lets me manually replace with new filament.

I tried printing again and faced the same message, this time I was keeping an eye on it and removed the filament before I got the heater timeout message. I removed the old filament and put in a fresh reel but when I press yes it doesn't proceed with the print

### Steps to Reproduce

1. Run out of filament
2. Replace filament
3. press yes

**Expected behavior:** heat the extruder and proceed with the print

**Actual behavior:** Hear the sound the printer makes when you select an option but it doesn't proceed with the print.

#### Additional Information

Bin file for the printer firmware I used
https://drive.google.com/file/d/1wlA83Kv4CRxz19cyr3tRvvxXg8McWM-l/view?usp=sharing

video of the issue
https://photos.app.goo.gl/51Fe5RmG9XnymSe77

picture of heater timeout
https://photos.google.com/u/1/photo/AF1QipPLKXjakY_T--VwEcCwbwzzwRR_PRPbVI6DztOv

InsanityAutomation, 2021-07-12:
It will typically take some time reheating before allowing you to continue. I checked the same build on a machine here this morning and was able to resume, though the message could be clearer that it is waiting for temperatures.. Ill add the temperature readouts on the screen so reheating is a bit clearer in the next revision. Can you confirm if waiting a few minutes for temperatures to rise will allow you to resume?

cai876, 2021-07-12:
Hi, Thanks for checking this out and your reply. I have tried waiting a few minutes and the hotend did heat up and I was able to restart my print. I think a temp readout and some simple instruction would definitely be a great addition to the firmware!

InsanityAutomation, 2021-09-10:
https://youtu.be/E0l_D9F_7BU - For the next update

closed 2021-09-10 by InsanityAutomation