InsanityAutomation/Marlin #214: Can't compile for E3 Mini V2 (2.0.9_DW7.4) Ender 5 Plus

quarky42, 2021-07-18:
Using the latest 2.0.9_DW7.4 pulled down today

### Description

`
warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=0 to SConscript calls.
Missing SConscript 'buildroot\share\PlatformIO\scripts\STM32F103RC_SKR_MINI.py'
`
When I manually add that file back in from the previous version of TM3D firmware, I get a new error:

`
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\Servo.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:934:5: error: static assertion failed: ADVANCED_PAUSE_FEATURE is required to use M600 with FILAMENT_RUNOUT_SENSOR.
     static_assert(nullptr == strstr(FILAMENT_RUNOUT_SCRIPT, "M600"), "ADVANCED_PAUSE_FEATURE is required to use M600 with FILAMENT_RUNOUT_SENSOR.");
`
When I add the -DADVANCED_PAUSE_FEATURE flag (I'm not sure why I have to add this since I'm already telling it that I'm using the -DEnder5Plus in the build flags) I now get this error:

`
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2600:4: error: #error "Please select only one LCD controller option."
   #error "Please select only one LCD controller option."
`
I want to use the stock touch screen with the TFT port on the SKRMiniE3V2 board.   I already do something just like this with the SKR14Turbo board and the touch screen on my CR10SPro and it works great.   

I've tried disabling the ORIG_LCD that is under the SKRMiniE3V2 section near line 349 in Configuration.h.  I still get that error.  I've tried adding the -DFORCE10SPRODISPLAY flag and removing it.    Same error.

I know you have reasons to hate the SKR Mini E3 V2, but you mentioned that you added support for it before.  Are you taking support for it back out?  I don't want to shell out another $45 for something that was supposed to be a drop-in replacement, but I don't want to give up on your firmware.  Your firmware works absolutely wonderfully on my other printer with SKRV14Turbo + CR10SPro.  I even got my stock touchscreen display to work easily with the FORCE10SDISPLAY flag over on that other printer / controller combo.

### Steps to Reproduce

1. Download
2. Create a section in PlatformIO for this printer configuration:

`:
[env:E5P_BIL_SKRE3V2]
# This is based on the STM32F103RC_btt settings above for the SKR E3 V2.0 Board
platform          = ${common_stm32f1.platform}
extends           = env:STM32F103RC
extra_scripts     = ${common.extra_scripts}
  buildroot/share/PlatformIO/scripts/STM32F103RC_SKR_MINI.py
build_flags       = ${common_stm32f1.build_flags} -DSKRMiniE3V2 -DMachineEnder5Plus -DHotendAllMetal -DBedDC -DADVANCED_PAUSE_FEATURE
  -DDEBUG_LEVEL=0 -DSS_TIMER=4
monitor_speed     = 115200
`

3. Try to build it.  Get that same error about the missing py script.   So I added STM32F103RC_SKR_MINI.py file from the previous version of the TM3D firmware.   Then I got the error about the ADVANCED_PAUSE_FEATURE, so I added that into the build_flags.  Now I'm getting the display already defined error.

**Expected behavior:** 
I expect it to compile a firmware for this printer configuration.  I've managed to add the missing file and the -DADVANCED_PAUSE_FEATURE flag, but I still keep getting more errors than I would expect.  I'd expect it to build just like my SKR14Turbo + CR10SPro configuration just by using a handful of build flags to merge the two together.

**Actual behavior:** 
It doesn't build.

When I try to build my existing SKR V1.4 Turbo configuration using a similar method above (copying the LCP1679 section and adding my CR10SPro build flags for that printer config) I get a compiled firmware that works great overall.

#### Additional Information

To be clear, I am now stuck at a point where I am getting this error after all the other errors have been solved (I think):

In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2600:4: error: #error "Please select only one LCD controller option."
   #error "Please select only one LCD controller option."

If I comment out the ORIG_LCD define that is in the SKRMiniE3V2 section around line 349 of Configuration.h I still get the same error message.   I am using the -DFORCE10SPRODISPLAY flag with my SKR14Turbo and CR10SPro printer and it works great.  On that printer the stock touchscreen is connected to the TFT port just like I have it wired up on this SKRMiniE3V2.   This ought to work.  Something is forcing a display type somewhere that I don't want.    I momentarily / temporarily also commented out the FORCE10SPRODISPLAY that is part of the:
"#if ANY(MachineCRX, MachineCRXPro, MachineEnder5Plus, MachineCR10SPro, MachineCR10Max, MachineEnder6)" logic just after line 485 (ish).   This finally let me get a compiled firmware file, but I don't expect this to work.   It seems like the assumption is that if you're using the SKRMiniE3V2, that you'll be using the ORIG_LCD, but I think I've proven with the SKR14Turbo board and the CR10SPro that the 10SPro style touchscreen does work with the TFT port that is on SKR boards.  What has to be changed / added to get the SKRMiniE3V2 to use the TFT port and the stock CR10SPro style touchscreen?

Can you please help me get the SKRMiniE3V2 with the stock touch screen connected to the TFT port working?  I've tried looking through the SKR14 Pins file to figure out what pins are used for the TFT, but the comments don't make it clear to me which is the TFT port and the BTT schematic is just as unclear for the TFT port (at least to me.)

quarky42, 2021-08-29:
The stock touch screen should be able to work with the SKR Mini E3 V2 board.   It's the same touch screen as the CR-10S Pro and this board has the TFT port just like the SKR v1.4 Turbo has.  The pin assignments and the logic that allows for the touchscreen to work with the CR-10S Pro / SKR v1.4 Turbo combo could be ported over to the Ender 5 Plus and SKR Mini E3 V2 combo please?

This is the Pinout of the touchscreen itself:
https://imgur.com/gallery/rcrRf0u

The RX goes to the TX on the controller TFT port.  The TX goes to the RX on the controller.  5v to 5v and Gnd to Gnd on the TFT port. This works great on the CR-10S Pro with the right flag set.   Need to support that FORCE10SPRODISPLAY flag on this combination, please.

I tried looking at it, but I couldn't tell which pins assignments were being made in the working combination and I think the pin names on the Mini E3 V2 are different too.

quarky42, 2021-08-29:
Ok, I was finally able to make some progress on this issue and I can get the stock Ender 5 Plus Touchscreen talking to the SKR Mini E3 V2.0.
I used information I found out from here where they have configurations for it already:
https://github.com/DodgeDeBoulet/Marlin-Deprecated/blob/E5Plus2.0.7.2/

I used info from their platformio.ini to see what flags were being set and then what was set in the configuration.h due to those flags.   I understand that some E3 V2.0's are not reliable for 512k memory and should only be used with 256k / limited features.   I'll do some testing, and see if I am able to use more.

I also saw how they are setting numerical values values using escaped \" which will come in very handy for me, too.

To get to this point I had to comment out the SERIAL_PORT 2 line and add LCD_SERIAL_PORT 2 and the 115000 baud rate.   I think commenting out SERIAL_PORT 2 disables the USB port, but I am still learning here.   I did see a note over on the description of the teaching tech video https://www.youtube.com/watch?v=mtCz_-2zvZo that says: "Tip from viewer Dale for helping this board connect easily to Octoprint when using a BTT TFT as well. Add this line to the "Additional Serial Ports" in settings: /dev/serial/by-id/usb-LeafLabs_Maple-if00"    I  need to investigate this to see if this will let me have my USB and TFT Port / stock touchscreen display working at the same time.

InsanityAutomation, 2021-09-10:
> Ok, I was finally able to make some progress on this issue and I can get the stock Ender 5 Plus Touchscreen talking to the SKR Mini E3 V2.0.
> I used information I found out from here where they have [...]
I wouldnt try, most severe issues with overflashed mcus were full hangs with heaters locked on full blast.

> I also saw how they are setting numerical values values using escaped " which will come in very handy for me, too.
> 
> To get to this point I had to comment out the SERIAL_PORT_2 2 line and add LCD_SERIAL_PORT 2 and LCD_BAUDRATE 115000 I think commenting out SERIAL_PORT 2 disables the USB port, but I am still learning here. I did see a note over on the description of the teaching tech video https://www.youtube.com/watch?v=mtCz_-2zvZo that says: "Tip from viewer Dale for helping this board connect easily to Octoprint when using a BTT TFT as well. Add this line to the "Additional Serial Ports" in settings: /dev/serial/by-id/usb-LeafLabs_Maple-if00" I need to investigate this to see if this will let me have my USB and TFT Port / stock touchscreen display working at the same time.

Youll see the serial port 2 options in configuration.h do the same for each board that is already supported. Thats the port being used for the screen, where the BTT display needs it in serial terminal mode instead.

quarky42, 2021-09-19:
Closing issue for the E3 Mini V2.  I had to fix a line in the Configuration.h to get the E3 Turbo to work with the Stock Touchscreen via the TFT port:   

The line that says:
#elif ANY(OrigLCD, MachineCR10Orig, MachineEnder3Pro422, MachineEnder3Pro427, MachineEnder3Max, SKRMiniE3V2, SKRE3Turbo) && DISABLED(GraphicLCD)
should replace the && DISABLED(GraphicLCD) at the end with:
&& NONE(GraphicLCD, FORCE10SPRODISPLAY)

Because without that, the code tries to define the CR10_STOCKDISPLAY and then it complains that there are too many displays defined. 

I'll open a new issue for this, but I've got some more work I need to do to make sure everything else is okay with the printer.

Some more comments in the PINS file would really help.

closed 2021-09-19 by quarky42