InsanityAutomation/Marlin #215: Ender 6 W/ BL Touch G28 (Home All Axis) Causes Thermal Runaway Protection to trigger

AccaliaDeElementia, 2021-07-23:
### Description

Using firmware_Ender6_BLT_DW7.4.bin.zip from 38fd3782f9040660a8ed8f4ac41109e6c659cb14 on Ender 6 with BL Touch Triggers Thermal runaway protection on G28 (Home All Axis) on Creality Ender 6 with v4.3.1 mainboard.

Alternatively, occasionally it will not trigger runaway protection, but will instead successfully home with the print bed anywhere from 20 to 300ish mm away from BL touch probe. This is rare compared to tripping thermal runaway protection and appears to be more common when the bed and hotend are at room temperature and not currently heating.

Issue did not exist on Ender-6-V1.0.4.9-BLTouch.bin firmware downloaded from Creality's official website, and upon reverting to that firmware without any other changes to the printer issue disappeared, however the other issues associated with using stock Creality firmware, particularly with Octoprint, reappeared... as to be expected.

### Steps to Reproduce

1. Boot Printer and complete a print
2. Clear print bed to begin a second print
3. Start second print
4. Become confused as your printer trips thermal runaway protection or succeeds to home without the bed ever touching the bl-touch probe

**Expected behavior:** Printer should successfully home for subsequent prints, as it does on the official 1.0.4.9-BLTouch firmware from creality

**Actual behavior:** Printer must power cycle between prints in order to reliably home prior to the beginning of prints

#### Additional Information

Log from octoprint of printer tripping thermal protection when homing after previous test print
[printer-log-thermal-protection-trip.txt](https://github.com/InsanityAutomation/Marlin/files/6870761/printer-log-thermal-protection-trip.txt)

notfixingit3, 2021-08-13:
I can confirm, I get this randomly (every 3-4 prints when done in a row without power cycle) when starting a new print. Screen just displays hardware fault, homing failed.

InsanityAutomation, 2022-02-26:
Filtering has been enabled in recent builds, which lowers accuracy however prevents the false triggers that were causing the homing failed errors.

closed 2022-02-26 by InsanityAutomation