InsanityAutomation/Marlin #216: 'Trex_2.0.x_Devel' branch - duplicate mode - 2nd extruder does not work

kvnper, 2021-07-29:
Using 'Trex_2.0.x_Devel' branch and T-Rex 3 (upgraded from 2)

When put in duplication idex mode, the 2nd hotend does not extrude at all.

Have not tested with mirror mode, could be the same issue there as well.

InsanityAutomation, 2021-09-10:
There are upstream issues that broke idex awhile back, and day job has been too overwhelming for me to get to it... I finally got the machine back together and up to date so hopefully ill get this resolved soon for everyone.

kvnper, 2021-09-12:
Thank you. Your personal and work life is far more important so please
don't stress about it.

InsanityAutomation, 2021-12-23:
I forgot to close the issue here as well, but last month i made it through most of the upstream IDEX issues.