InsanityAutomation/Marlin #217: I just updated my CR 10S Pro to the latest firmware 2.0.9_DW 7.4 with the screen files and notice that when I turn on the computer the screen has a message SD Init Fail.  What is this?

rlajordan, 2021-07-29:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

undert03, 2021-09-07:
I am having the same issue

CR10S Pro flashed successfully with 10SPro_BIL_DW7.4.hex.zip
Screen files: TM3D_DWINCombinedScreens_V4.7z

After boot it says "SD Init Fail"

Also, the normal sounds that happen out of the speaker sound all static/distorted.

All worked just fine under 1.1.9_B7 Firmware

Video of issues 
https://www.dropbox.com/s/xnv4m4gfpul04me/CR10sPro-2.0-DW7.4-bugs.MOV?dl=0

InsanityAutomation, 2021-09-10:
@undert03 are you perchance on a Mac? The default archiver is known to corrupt the wav files. See the note in the readme. Please use 7zip.org to extract the files.

@rlajordan If there is no SD card present, it will fail to pull a file list. This extra forced check was added due to an increasing number of users reporting broken SD detect pins, so were attempting to force the card read at bootup.