InsanityAutomation/Marlin #218: Feature request: Creality CR-200B support

mcecchi, 2021-07-31:
Add support for Creality CR-200B enclosed 3d printer

InsanityAutomation, 2021-09-10:
If I can get one I will definitely add it. @tinymachines-3d has offered an Ender 7 and CR10 Smart first, and I have code to reference for the CR5 which just leaves the Sermoon D1 and this in the cold for now.

piotrkpaul, 2021-12-13:
any update on this one?

InsanityAutomation, 2021-12-23:
Not yet, CR10Smart is here now, CR5 arrives next, then the CR200B will be behind it. We did get the new revision screen in it working with the Sermoon / Ender 7 stuff recently so it should be very fast with the base machine functionality only needing to be sorted.

edriyin1, 2024-02-03:
Is there any updates on this?

jaymcd, 2024-08-15:
I also would love to see the CR-200B added.

jaymcd, 2024-08-26:
There was some work done here a while back - love to see if this could be updated for current Marlin - https://github.com/pienjo/Marlin-CR200b

xforce2099, 2024-10-05:
Hi I recently bought a CR200b and noticed that it has a very sad lack of resources for thinkerers that would love to extend the printers functionality. (I want to enable linear advance on mine and have truble doing so) I'm a software dev and would love to make this feature request possible, but I have zero ecperience with this kind of stuff, so if someone would have time to assist me I would love to make this happen.