InsanityAutomation/Marlin #221: Ender 6 - Z-Offset Value Issues on LCD Display

Tw0bit, 2021-08-18:
### Description

Ender 6 (https://github.com/InsanityAutomation/Marlin/blob/CrealityDwin_2.0/Hex%20Files/firmware_Ender6_DW7.4.bin.zip):

- Z-Axis Compensation on Print Screen always shows 0 even when you tap + or -
- Manual Mesh process shows only 1 decimal value (i.e. 0.2) but the steps are in 0.025 - so 4 steps to change the shown value

### Steps to Reproduce

Print Interface
1. Print a file, adjust the Z-Axis Compensation - always shows 0

Level Interface
1. Start the Manual Mesh Process
2. The shown value box is 1 decimal value but require 3 to level with the 0.025 steps

**Expected behavior:** [What you expect to happen]
Value shows as epected

**Actual behavior:** [What actually happens]
Value not showing

Tw0bit, 2021-08-23:
How would i make/save a +0.08mm offset on this firmware? Each print i have to adjust Z by 4 + clicks.

kaodra, 2022-04-24:
Also experiencing this, setting the z-offset lower or higher just causes it to remain 0 on the display. It does change, but if you try to save it, since it displays 0, it also saves 0. So nothing changes or saves. Kind of a bare minimum requirement for printing....

SplinMS, 2023-06-16:
Z-Axis Compensation on Print Screen always shows 0 even when you tap + or -

burockoz, 2023-09-17:
the z offset value is limited to -9, no matter what I do, I always stay 4-5mm away from the bed.