InsanityAutomation/Marlin #235: Martin with configuration for a BIQU H2

onkel645, 2021-11-14:
Hello,
i have a question.(Sorry for my Bad english) ;-)
I have an Ender 6 with the DW7.4.3 Firmware. Now i made an Upgrade with the BIQU H2 Extruder.
I change the Rotation distance for the Extruder. This was easy, but for the the following settings i need a „new“  configuration.h for my Ender. 
Can anyone made a new 7.4.3 version for me with these Settings:
[stepper_y]
position_endstop: 245 # moves y 0 position to keep nozzle from traveling too far forward off bed.
position_max: 245
...
[safe_z_home]
home_xy_position: 130, 122.5 # accounts for new y position_endstop
...
[bltouch]
x_offset: -34.2
y_offset: 0
z_offset: 0 # temporarily reset to 0 to prevent smashing nozzle into bed.
...
[bed_mesh]
mesh_min: 0, 0
mesh_max: 225.8, 245 # position_max + offset to stay within travel limits.
...
[extruder]
rotation_distance: 3.4334

Thanks an Greetings.

subhead, 2021-11-25:
I would like to ask the same. I have switched all my Ender 6 to the BIQU H2. Is there a way to obtain the configuration files that we can compile our one  version with the needed configuration?

Thanks in advance!

subhead, 2021-11-25:
Is this configuration (https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/Ender-6) the same that was used to build 7.4.3/7.4.4?

InsanityAutomation, 2021-12-23:
The 743 configs are all here - https://github.com/InsanityAutomation/Marlin/releases/tag/DW743

The early H2 extruders had severe gear mesh issues. I have a new one now that is supposedly fixed, but I havnt had time to mess with it.

What mounts did you use? Maybe ill drop it on the E6 to see how it goes.

subhead, 2021-12-26:
Thanks for the configs.

i use this mount für my h2 https://cults3d.com/de/modell-3d/verschiedene/ender-6-biqu-h2-extruder-mount-bwi

closed 2023-02-17 by InsanityAutomation