InsanityAutomation/Marlin #238: TM3D_DWINCombinedScreens_V6.7z  sound problem

MorganTr44, 2021-11-29:
### Description

strange sounds on CR10S PRO VE with TM3D_DWINCombinedScreens_V6.7z and 10SProV2_BIL_DW7.4.4.hex firmware

### Steps to Reproduce

1. install firmware
2. install screen binary
3. reset settings and restart

**Expected behavior:** [What you expect to happen]

known sound (as find in .wav files

**Actual behavior:** [What actually happens]

unknown sounds

#### Additional Information

closed 2021-11-29 by MorganTr44

rebelonion, 2021-12-07:
I am also having this same problem on CR-10S PRO V1 with TM3D_DWINCombinedScreens_V6.7z and 10SPro_BIL_DW7.4.4.hex firmware.

phehr2, 2021-12-07:
I am also having this same problem on CR-10S PRO V1 with TM3D_DWINCombinedScreens_V6.7z and 10SPro_BIL_DW7.4.4.hex firmware.