InsanityAutomation/Marlin #240: dw7.4.4 not  working with octoprint

turbodlr, 2021-12-08:
Octoprint connects fine at 250000 baud.. couple things. 
1.I can't send commands such as M503 through terminal, It shows nothing but wait commands scrolling.   
2. Ocotlapse reports printer not compatible with m114 command, 
3. starting print from octoprint does nothing to printer, octoprint says printing but printer never heats etc.   Am I missing something?

turbodlr, 2021-12-08:
Oh thisis on ender 5 plus btw

turbodlr, 2021-12-08:
update, in safe mode on octoprint I can send commands in terminal and i can kick off a print.

Tamas92, 2021-12-22:
For me it is working on 7.4.4 on CR-10S Pro V1 without any modification. I used the DW 6.2 before and upgraded today to 7.4.4.

Silverstar, 2022-12-05:
I can print on 7.4.6 on E5P with octoprint, but some stuff like EEPROM Editor loading gets stuck until I fake ack in the terminal tab of octoprint.
Same symptom and reason as in #288 - M503 ends with an incorrect line, and so octoprint gets stuck.