InsanityAutomation/Marlin #242: Screen Doesnt work properly Ender 7

BlauMink, 2021-12-17:
### Description
just installed the software on my ender 7, tried many times but i allways end up getting corrupted images when i open certain menus

### Steps to Reproduce

1. Flash the latest version of the Screen frimware
2. try to check software version, or refuel the printer or stopping the mottors, it either doesnt work or just shows a corrupted image

**Expected behavior:** to see the menu images working how they are suposed to, Aka normal functionallity

**Actual behavior:** cant disconect the motors and certain menus appear like corrupted images

#### Additional Information
i tried re-downloading the files, different versions and flashing the screen and printer several times with no success
im using the latest version of your firmware for the printer, 7.4.4, version with bl touch plugged in the z stop port
im also using 7-zip to unpack everything, and when inspecting the images themselves everything seems to be ok

images of some of the problems, also the "control motors" menu directly wont open
https://imgur.com/gallery/nbnsdQr

BlauMink, 2021-12-17:
i also just discovered that i cannot stop/abort prints

aaron-hentschel, 2021-12-21:
I have a very similar issue on my Sermoon D1 as @BlauMink has on his Ender 7.

It would also be great if the Z+ and Z- buttons could be swapped around on the Move screen since the Z axis as home at the top of the chassis (move the Z+ to the bottom of the screen and Z- to the top)

InsanityAutomation, 2021-12-23:
Ive been trying different archiver / browser combinations and havnt found a rhyme or reason yet to when it decides to corrupt the flashed ICL... I have one machine that i havnt seen a corrupted display on yet, and another that seems to be 50/50. I did get an update to the dgus software from 8.2.10 to 8.2.14 yesterday that im hoping will be more stable... Ill be working on it over the holiday break a bit.

InsanityAutomation, 2022-02-26:
update resolves screen corruption issues.

closed 2022-02-26 by InsanityAutomation