InsanityAutomation/Marlin #243: 10SProV2_BIL_DW7.4.4 Move Button changes language

Simdev33, 2021-12-21:
Hello!

On my CR10S PRO V2 after upgrading the firmware to 7.4.4, the Move button changes the language to chinese. It doesn't open the move options, only the text changes to chinese and that's it. After that I can change it back to English by pressing the tool button and it immediately offers the option. 
Is it normal behaviour or I did something wrong?

Tamas92, 2021-12-22:
I have the same issue on CR-10S Pro V1.

phaeton82, 2021-12-22:
I have the same issue

Depdog028, 2021-12-24:
Same issue on Ender 6, reverted to 7.4.3

Kaelum, 2022-01-05:
They changed the interface with the display when 7.4.4 was released, thus you need to update the display to the latest version, which is 6.7 for the Ender 5 Plus (TM3D_DWINCombinedScreens_V6.7z).  I don't know if it's the same for other machines, as the Ender 5 Plus had it's own specific firmware.

soloroboto, 2022-01-06:
Updating to the 6.7 LCD files fixes the "swap to Chinese" error (I just went through exactly this). 
But now I've lost most audio - button clicks no longer make a sound at all, and only some confirmations get audio.

InsanityAutomation, 2022-02-26:
Update to 745 resolves this issue

closed 2022-02-26 by InsanityAutomation