InsanityAutomation/Marlin #244: Babystepping with 0.1mm steps instead of 0.25mm steps

Ericvf, 2021-12-21:
I'm aware that there was a discussion about this and that it was decided that 0.25 steps should be enough for most people. It must be the cold weather but I find myself using babystepping a lot these days when dialing in filaments. My satin filaments usually want a little more proximity to the bed and therefore I give them nudges. 

I would however really like the 0.1 steps back. It doesn't matter to me that I have to use more button presses. If I am correct there is no feature toggle for this. Can you point out which part of the code is responsible for that? I've looked at some of the code but it isn't apparent to me where this happens. 

Also because of the rounding sometimes you can go from 1.20 down to 0.90 and back up to 1.30 (or something weird like that). 
Please advice, I have no issue with patching this in my own fork

Kaelum, 2022-01-05:
+1 on this, as it doesn't really make sense to arbitrarily set it to some random value like 0.2 (in TM3D_DWINCombinedScreens_V6.7z) or 0.25 in the OPs case.  There is no advantage to be gained, just functionality lost.

InsanityAutomation, 2023-02-17:
The adjustment screen is .02mm for fine tuning, and .1 on the rough leveling screen. I cant have 200 versions to make everyone happy, so I keep as many as I can and everything needed to modify both the screen and code is here for the rest.

closed 2023-02-17 by InsanityAutomation

Ericvf, 2023-02-18:
> The adjustment screen is .02mm for fine tuning, and .1 on the rough leveling screen. I cant have 200 versions to make everyone happy, so I keep as many as I can and everything needed to modify both the screen and code is here for the rest.

Could you at least explain where this value can be changed so I can adjust my own fork. 

I've been very happy with these firmwares and I would like to thank you again for all the hard work. 

Unfortunately the later releases have been troublesome for my CR-10s Pro v1. I still have audio issues, this stuff with baby stepping and some issues with octoprint bed leveling.

InsanityAutomation, 2023-02-18:
The audio is a symptom of the newer toolset... Since machines showed up using DGUS 2, 3.5, 4.0, 4.1 and 4.5, we needed to go to the newer toolset that would make binaries for the LCD compatible with all of them. It does butcher the sound on the V2 OS screens, but thats the only caveat im aware of, and has been basically ignored as I believe everything currently shipping has the newer OS displays.

To change the leveling screen offset, 
[image]
 in Creality_Dwin.cpp

To change the fine adjustment, that needs to be done in the LCD software, changing the step size on the button (DGUS Toolbox)

[image]

The octoprint issue i was waiting on the upstream PR to get merged, but Scott created a permissions issue editing on a Mac I couldnt fix from win32, and he just ignored it so I closed it. Ill go back and fix it here as I play catchup. Those on discord have the story of where ive been the last few months, but been a rough bit and finally digging out.

Ericvf, 2023-02-27:
Thanks for the clarification and the details for the Babystepping, that will help a lot!