InsanityAutomation/Marlin #245: Ender 5+ w/silent board. Push move button on touchscreen and language changes to Chinese. Doesn't allow me to move any axis from TS

deke77, 2021-12-27:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Kaelum, 2022-01-05:
You need to update to the newer DWIN_SET package for the display, as they changed the interface with the release of 7.4.4.

psiess, 2022-01-17:
> You need to update to the newer DWIN_SET package for the display to TM3D_DWINCombinedScreens_V6.7z, as they changed the interface with the release of 7.4.4.

Hi @Kaelum.  Sorry for the n00b question, but where do I download the updated DWIN_SET package?

Kaelum, 2022-01-17:
@psiess change the branch to `CrealityDwin_2.0` and you'll see the `TM3D_DWINCombinedScreens_V6.7z` file at the root of the repository.

psiess, 2022-01-17:
@Kaelum got it. Thank you!

closed 2022-02-26 by InsanityAutomation