InsanityAutomation/Marlin #256: Support for Ender 6 on 4.2.7 Board

mtregard, 2022-02-16:
### Description

The Ender-6 comes with 4.3.1 Boards that have cheap stepper-drivers for Z and E. Especially the E is driving me crazy when printing with retractions. I tried with a 4.2.7 Board that has all 2225 drivers, but stock firmware and also yours will not move the steppers. Same board with Ender 3 V2 firmware moves all steppers, but naturally config is all wrong.  

CAN YOU PLS MAKE A BUILD FOR THIS SETUP? Or at least give me some hints as to what I would need to change to compile it myself?!!!

mtregard, 2022-02-17:
For all of you looking for a solution. Build yourself using Platform.io.

1. in **configuration.h** change 

```
  #elif ENABLED(MachineEnder6)
    #define MOTHERBOARD BOARD_CREALITY_V431
```

to

```
  #elif ENABLED(MachineEnder6)
    #define MOTHERBOARD BOARD_CREALITY_V427
```

2. Build Ender 6 Image to your liking
3. **_IMPORTANT:_** swap X&Y Stepper cables on the board (it seems Ender 6 is COREYX not XY ;)). I believe this could be changed with config as well, but this is my first look into Marlin source-code so I skipped that for now.

mtregard, 2022-02-17:
So it should be fairly easy to add a build-config for this very much desirable scenario...

InsanityAutomation, 2022-02-26:
Just to be sure I understand the reasoning behind the board change, is this driven by cost / availability of replacements? Not difficult to add at all as a predefined configuration though.

mtregard, 2022-02-28:
Thanks for getting back to me. It's purely a matter of noise from the stepper-drivers for Z and E. I believe not all 4.3.1-boards are affected by this, but I also know that I'm not alone. I tried 3 such boards from different sources and they all had cheap drivers on them. When printing with lots of retractions it's a real pain.

houseofbugs, 2022-03-14:
The 4.2.7 is not a direct replacement for the 4.3.1. If you are swapping to the 4.2.7 then you should have the knowledge to do so, including firmware changes. This is not an issue, but a user problem so it should not be present in the Github issue queue.

mtregard, 2022-03-14:
@houseofbugs : I agree that this is not at all an issue, but I did not know any other means of getting attention. Not everyone who can swap a board is capable of doing firmware changes. The Ender 6 with a BIQU H2 and some other mods CAN be a very capable and silent machine at a more than acceptable price point. Would be nice if others could simple download your excellent firmware and continue printing....

houseofbugs, 2022-03-14:
> @houseofbugs : I agree that this is not at all an issue, but I did not know any other means of getting attention. Not everyone who can swap a board is capable of doing firmware changes. The Ender 6 [...]

That is not @InsanityAutomation problem if you're too incompetent or lazy to make a non standard board work with a specific machine.

The issue queues on GitHub are not for your own personal tech support. Stop doing it. 

If you're not willing to make modifications to the source code for your particular printer then you should probably not be doing those types of changes and should pay someone to do it for you, not demand it on GitHub issue queues because you want it done for free because you're cheap.

closed 2023-02-17 by InsanityAutomation