InsanityAutomation/Marlin #262: CRX Pro

Looping-39, 2022-03-23:
Good morning, 

3 issues noted on the CRX and CRX pro with the latest 7.4.5 :

1: This is the parking point at the change of filament when using the 2 extruders.
The nozzle moves outside the tray until the end of YX (about Y-325 X+325). 
After heating the extruder and changing the filament, the nozzle returns at the beginning of YX (Y 0 X +325) always outside the tray.
Then it heats up again and then moves to print a priming tower for example and then goes to the room.

2: Screen: degree figures overlap

3: palpation BL touch: random, but on the screen the values do not correspond to the measured points. 

Thanks. 

P.S. I'm a noob

InsanityAutomation, 2022-03-29:
With UBL builds, the mesh index reported during probing and the actual mesh position dont align. After recalling from eeprom or a power cycle however they are loaded to the correct points. Its an issue in the UBL code upstream and not something in the scope to fix here.

Parking routine here should be a single point, then  wipe move for those with a wiper box. That full behavior has been modified several times by others since we originally wrote the automatic parking and material swap. If the people who changed the behavior dont make it back to fix it, itll likely fall back to me to pick up the pieces yet again.

Screen text overlap has had some cleanup, the back end platform update drew things very differently after the update. Its definitely a low priority task though!

Looping-39, 2022-03-31:
Thank you for all these explanations.  good continuation.

closed 2022-03-31 by Looping-39