InsanityAutomation/Marlin #267: Requesting access to configuration files

jornamon, 2022-05-13:
Instead of using a compiled file for a certain combination of hardware/options, I would like to download the configuration files used for that build, so I can use them as a “safe” starting point, make small adjustments and then compile the firmware.
I have searched the repo, but it seems that the configurations files for a particular build are not saved. Could I use your scripts to generate them or something like that?

Thanks!!

Ericvf, 2022-07-04:
You need to look at the Platformio.ini. 

Those are the configurations that they probably use for their precompiled files. 
```
[env:10SPro_BLT_BIL]
platform      = atmelavr
extends       = common_avr8
board         = megaatmega2560
build_flags = ${common.build_flags} -DMachineCR10SPro -DHotendAllMetal -DBedDC -DABL_BLTOUCH
```

Those build flag directly correspond to the `#define`'s inside the `configuration.h`. If you want to rebuild the firmware, try looking at that.

InsanityAutomation, 2023-02-17:
As the comment noted, platformio.ini calls out everything for each corresponding config.

closed 2023-02-17 by InsanityAutomation