InsanityAutomation/Marlin #274: CR-10 MAX Bed Size defined incorrectly

DragonSilverSky, 2022-07-03:
Installing the Hex file for  DW7.4.6 onto my CR-10 MAX, lots of improvements but the defined bed size is 470x470x470.  Believe the correct is 450x450x470.

Thank you for all you do!

igorka48, 2023-09-22:
+1 to question. 
Any solutions?

DragonSilverSky, 2023-09-22:
Hey Igorka48,

My CR-10MAX has been offline since I found a massive 'cupping' in the center of the head bed.  I have sourced a new heat bed and 20x10 extrusion to support the center and prevent cupping.  I have yet put it all back together.  Once I do, my plan is to edit the firmware lines directly and upload to the printer.

I would suggest doing that yourself.  Good Luck, and Cheers!

igorka48, 2023-09-22:
@DragonSilverSky edit only bed size parameters will be enough? or should edit some other params?