InsanityAutomation/Marlin #275: 10SPro_BLT_BIL_DW7.4.6 having various issues

Ericvf, 2022-07-04:
### Description

After a long while I decided to update the firmware. The previous version was 7.3.1, which I compiled and flashed myself. Everything in this version was working fine for me: BLTouch, Babystepping, Bedlevelling etc. 

### Steps to Reproduce

1. Open VSCode
2. Clone InsanityAutomation/Marlin
3. Checkout Creality_DWIN_2.0
4. Switch to project environment: 10SPro_BLT_BIL
5. Build
6. Upload
7. Extract TM3D_combined_XXX_V7.zip and copy DWIN_SET to an SD card
8. Insert SDCard into screen board and turn on machine. 
9. Wait for update to finish

**Expected behavior:** 

Everything works. 

**Actual behavior:**

1. Homing is not center of the buildplate but one inch to the back of the buildplate.
2. Measuring for Auto Bed Levelling is also offset one inch to the back. Nearly touching the edge of the buildplate.
3. Audio is played wrongly. Confirm sound is some long startup audio file. 
4. UI elements such as delimiters, slashes and others are not present
5. Graphics looks compressed

#### Additional Information

This is my `configuration.h` file. Very little changes made:
https://github.com/Ericvf/Marlin/commit/00f39cfc2f6371d5a67ebfb22c18806b2c9ba076

Ericvf, 2022-07-04:
Printing seems to work ok. Interestingly it does print at the correct location on the bed. I've also noticed the UI no longer supports changing the print temperature during print. Why is that?

IXI3DBE, 2022-07-05:
hello, @Ericvf
[image]

It seems to me that the procedure is described in the Readme file.
Just press once on the touch screen where the nozzle and bed temperatures are displayed.
Press once to exit the sleep mode screen and a second time to display the numeric keypad.
that's all

Ericvf, 2022-07-05:
Thanks @IXI3DBE. Good point, I did not know what. To exactly which manual are you referring to BTW? 
Just tried it and indeed the numeric keypad pops up there. Nice!

IXI3DBE, 2022-07-05:
hello,
for once I read the manual! (it's not often)
[image]

Ericvf, 2022-07-05:
Allright so it turned out I needed to reset the eeprom. Forgot to do that. That solves the homing/bed levelling position. The only issue I have now is the audio, which isn't a huge problem for me. I can wait those fixes out as I read about them in other topics.

Ericvf, 2022-07-05:
Thanks again for the responses @IXI3DBE.

closed 2022-07-05 by Ericvf