InsanityAutomation/Marlin #276: "Finish print" button and "Print aborted"

Ericvf, 2022-07-11:
### Description

When I finish a print the printer is always waiting to me to press Finish Print. When I do it looks like the print becomes aborted. That is the message that I see in the screen. 

When using the Octoprint Bed Levelling Visualizer I have a similar issue. After probing the printer will wait for me to press Finish Print. But when I do that, Octoprint never receives a mesh. I'm not entirely sure if the Finish Print button is related to the Bed Levelling Visualizer issues. 

Does anyone have any idea? I'm using a standard configuration on my CR-10s Pro (10SPro_BLT_BIL)

djqpaz, 2022-08-21:
I'm having the same issues

Ericvf, 2022-08-21:
Thanks for commenting. Still haven't found a way to change this behavior. Hopefully one of the developers can comment on this soon!