InsanityAutomation/Marlin #280: Ender 7 config not found..

simrim1, 2022-08-28:
Hi,
First of all I am sorry if I am posting on some unrelated issue. I am a noob and learning compiling Marlin for my Ender 7 printer. Please move this post to appropriate location if need be.

I came across this [post](https://github.com/MarlinFirmware/Configurations/issues/602#issuecomment-954677721)  on Marlin firmware issues page where @InsanityAutomation has mentioned that a config file for Ender 7 has been added to his fork but I couldn't find it.

Can someone, please, point out where I can find the config file for Ender 7?

Thanks

rbora13, 2022-09-05:
Hi,
You just have to :
1. download Insanity Automation Marlin sources
2. install VScode and platformio and open Marlin sources 
3. in the platformio.ini file, you remove all default_envs except Ender7. So you should have: default_envs = Ender7
4. compile sources
5. flash the bin file
Concerning the screen firmware, you simply copy the TM3D_Combined480272_Landscape_V7\DWIN_SET folder (unzip the TM3D_Combined480272_Landscape_V7.7z file) to a microSD card and insert the card in the screen's µSD card reader before switch on the printer.
Enjoy

enderbrother, 2022-12-11:
> Hi, You just have to :
>
> 1. download Insanity Automation Marlin sources
> 2. install VScode and platformio and open Marlin sources
> 3. in the platformio.ini file, you remove all default_envs except [...]

I am utterly lost. I've been trying to figure this out so I can use the change filament command, but ended up freezing the screen on my ender 7

enderbrother, 2022-12-11:
I may need elaboration on those steps

tcs320, 2022-12-13:
Hi,

I’m a little bit lost as well and I’d have used vscode and Marlin before on another printer. As I understand it we should use the normal Marlin fork from Insanity Automation Fork on GitHub without replying the configuration files. 

But then as you @rbora13 describe I can’t find any lines default_envs to remove in the platformio.ini file there is just one. Is it enough to just edit this line and past Ender 7 into it and the the default Configuration files just work?

Dids, 2023-02-04:
Any updates on this? No matter which Ender 7 environment/config I try, I get errors one way or another. The more I attempt to fix, the more errors I get.

First I had to add the following to `ini/stm32f1.ini` to stop it complaining about the missing `common_stm32f1` env:
```ini
[common_stm32f1]
extends = common_STM32F103RC_variant
```

But now I get compilation errors, such as the following:
```
Building in release mode
Compiling .pio/build/Ender7/src/src/inc/Warnings.cpp.o
Compiling .pio/build/Ender7/src/src/lcd/extui/Creality/Creality_DWIN.cpp.o
Compiling .pio/build/Ender7/src/src/lcd/fontutils.cpp.o
Compiling .pio/build/Ender7/src/src/lcd/marlinui.cpp.o
Compiling .pio/build/Ender7/src/src/libs/BL24CXX.cpp.o
Compiling .pio/build/Ender7/src/src/libs/buzzer.cpp.o
Compiling .pio/build/Ender7/src/src/libs/crc16.cpp.o
Compiling .pio/build/Ender7/src/src/libs/nozzle.cpp.o
Marlin/src/inc/Warnings.cpp:798:4: warning: #warning "Input Shaping for CORE / MARKFORGED kinematic axes is still experimental." [-Wcpp]
  798 |   #warning "Input Shaping for CORE / MARKFORGED kinematic axes is still experimental."
      |    ^~~~~~~
Compiling .pio/build/Ender7/src/src/libs/numtostr.cpp.o
Marlin/src/lcd/extui/Creality/Creality_DWIN.cpp:783:6: error: default argument given for parameter 4 of 'void ExtUI::RTSSHOW::WriteVariable(uint16_t, const void*, uint8_t, bool, char)' [-fpermissive]
  783 | void RTSSHOW::WriteVariable(uint16_t adr, const void* values, uint8_t valueslen, bool isstr=false, char fillChar = ' ') {
      |      ^~~~~~~
In file included from Marlin/src/lcd/extui/Creality/Creality_DWIN.cpp:34:
Marlin/src/lcd/extui/Creality/Creality_DWIN.h:244:10: note: previous specification in 'void ExtUI::RTSSHOW::WriteVariable(uint16_t, const void*, uint8_t, bool, char)' here
  244 |     void WriteVariable(uint16_t adr, const void* values, uint8_t valueslen, bool isstr=false, char fillChar = ' ');
      |          ^~~~~~~~~~~~~
Marlin/src/lcd/extui/Creality/Creality_DWIN.cpp:783:6: error: default argument given for parameter 5 of 'void ExtUI::RTSSHOW::WriteVariable(uint16_t, const void*, uint8_t, bool, char)' [-fpermissive]
  783 | void RTSSHOW::WriteVariable(uint16_t adr, const void* values, uint8_t valueslen, bool isstr=false, char fillChar = ' ') {
      |      ^~~~~~~
In file included from Marlin/src/lcd/extui/Creality/Creality_DWIN.cpp:34:
Marlin/src/lcd/extui/Creality/Creality_DWIN.h:244:10: note: previous specification in 'void ExtUI::RTSSHOW::WriteVariable(uint16_t, const void*, uint8_t, bool, char)' here
  244 |     void WriteVariable(uint16_t adr, const void* values, uint8_t valueslen, bool isstr=false, char fillChar = ' ');
      |          ^~~~~~~~~~~~~
*** [.pio/build/Ender7/src/src/lcd/extui/Creality/Creality_DWIN.cpp.o] Error 1
======================================================== [FAILED] Took 3.80 seconds ========================================================

Environment    Status    Duration
-------------  --------  ------------
Ender7         FAILED    00:00:03.799
================================================== 1 failed, 0 succeeded in 00:00:03.799 ==================================================
```

InsanityAutomation, 2023-02-06:
Looks like you grabbed the bleeding edge development repo, not the release repo. Grab the standard CrealityDWin2.0 branch and there is both a precompiled bin as well as all the options used to build it flagged under the matching environment name in platformio.ini

Dids, 2023-02-06:
> Looks like you grabbed the bleeding edge development repo, not the release repo. Grab the standard CrealityDWin2.0 branch and there is both a precompiled bin as well as all the options used to build it flagged under the matching environment name in platformio.ini

Thanks! That's essentially what I ended up doing, and realized my mistake shortly after (edited the original post then).

The `CrealityDWin2.0` branch compiled just fine, and I went ahead and flashed the BLTouch version, as well as the `TM3D_Combined480272_Landscape_V7.7z` touch screen firmware.
However, I'm not entirely sure if the display firmware is supposed to have several larger issues/bugs, or if I managed to somehow corrupt it, as I'm seeing very bad artefacting on the leveling screen (lots of pink and illegible text), but also noticed that while touch events cause an audible beep, they rarely register and I may have to keep tapping buttons anywhere from 2 to 10 times, before they actually register. Lastly, when babystepping while printing, the Z offset very suddenly jumped from roughly -2.0 to +8.0.

All in all, I'm not entirely sure if these are quirks of the custom display hardware, issues Marlin side, or if there's been hardware changes in the Ender-7 lineup that are causing these irregularities, unless of course these are all already known bugs, as I do understand the massive undertaking of this project/repository and definitely don't expect any support, but more so curious to the status of things, where I could potentially help etc. :)

InsanityAutomation, 2023-02-17:
Ive seen that kind of artifacts from stock mac archivers corrupting files in the past. The serial issues came about when the upstream serial class was refactored to be smaller but dropped stdlib code for custom code. Buffers are overflowing. Only way ive seen to stop it completely is a specific firmware version for the screen itself, which would require re-calibrating the touchscreen as well. Ive got a couple other tricks I may try eventually to speed up processing, but for now at least...

As the original issue is gone, im closing this to clear up the issues thread but discussion can continue if needed.

closed 2023-02-17 by InsanityAutomation