InsanityAutomation/Marlin #290: CR10S Pro V2 bleeding and 2.0 yY+ is reversed causing bed homing issue

oioiandy, 2022-11-20:
-->

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Double077, 2023-01-04:
I have the same exact issue !!!!!
I have solved the reversed y-axis by reversing the wiring in the y-motor but the creepy girl is annoying did u manage to do something about it?

sharkbite86, 2024-04-22:
Y-axis seems fixed with the latest bleeding v7.4.9 on my CR10S Pro V2. Haven't had this issue. Screen button sound is still creepy though ha.