InsanityAutomation/Marlin #299: Ender 6 moving to front after stop print

justinkumpe, 2023-03-19:
### Description

If I stop the print while printing the extruder will move to the front then left hitting the frame. Home position should be back and right.

### Steps to Reproduce

1. Start a print
2. stop it mid-print

**Expected behavior:** [What you expect to happen]
When print is stopped/cancelled the extruder should home to back right.

**Actual behavior:** [What actually happens]
When print is stopped/cancelled the extruder homes to front left.

#### Additional Information

When a print completes successfully or home button is pressed the extruder homes to back right correctly just not when print is stopped/cancelled.

melp12345, 2023-04-10:
same problem on my ender 6 ,
however if i dont set up any bed levelling the print head does not crash into the door 
but soon as auto level if stored every time i stop the print it crashes 
something must being corrupt when it stores

famtory, 2023-06-01:
same my ender 6.

Goodsmileduck, 2023-06-03:
same happening for my ender 6

SplinMS, 2023-06-16:
same happening for my ender 6

bozmanx, 2023-08-10:
> Hi This is working for me: Configuration_adv.h Change: #define EVENT_GCODE_SD_ABORT "G91\nG1Z5\nG90\nM84\nM104S0\nM140S0" To: #define EVENT_GCODE_SD_ABORT "G28 X0 Y0\nG1Z5\nG90\nM84\nM104S0\nM140S0"
> 
> Ender7, but should be ok for Ender 6, just change your home position (G28 X? Y?) But should be the same, G28 - Auto Home

I upgraded my ender 6 firmware a couple days ago and I found this issue on my first print. Great find, so how do I fix this ? Do I need to recompile the code somehow or is there a way to edit it manually ? 

Is it possible to fix this by editing the end gcode ? Here is my End gcode, it clearly has the G91 command

G91 ;Relative positioning
G1 E-2 F2700 ;Retract a bit
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positioning

G28 X Y ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed

M84 X Y E ;Disable all steppers but Z

Any help would be greatly appreciated.

diymarcus, 2023-08-10:
> I upgraded my ender 6 firmware a couple days ago and I found this issue on my first print. Great find, so how do I fix this ? Do I need to recompile the code somehow or is there a way to edit it [...]

Hi, I am not a big specialist in Marlin firmware, but i think is not possible to add a Gcode in end printing Gcode, this bug is about forced stop, Will be better to edit Configuration_adv.h and recompile firmware. Good Luck

SWXTools, 2023-11-21:
Does anyone has a compiled Ender 6 binary (without BLT)? i don`t know what i have to do in Marlin.

Thanks @ all

SWXTools, 2023-11-24:
> Hi This is working for me: Configuration_adv.h Change: #define EVENT_GCODE_SD_ABORT "G91\nG1Z5\nG90\nM84\nM104S0\nM140S0" To: #define EVENT_GCODE_SD_ABORT "G28 X0 [...]

Here is a compiled Endstop-Version of the 7.4.7 with the recommended change.
There is also an original binary from Creality to downgrade, if you've actually installed the 7.4.7.
[https://drive.google.com/file/d/1M5MQHNa4GEI3HJ2Im3pAEoCO0GOa1nbb/view?usp=drive_link](https://drive.google.com/file/d/1M5MQHNa4GEI3HJ2Im3pAEoCO0GOa1nbb/view?usp=drive_link)