InsanityAutomation/Marlin #303: CombinedLandscapeDwin_TM3D_V8 DWprj.hmi

Ericvf, 2023-05-21:
I'm using CombinedLandscapeDwin_TM3D_V8.7z for my CR-10s Pro V1 and I would like to have the V8 DGUS project if possible. The V7 version was available previously and I edited that a bit (adjust baby stepping values). Miraculously the V7 (FW 7.4.6) DWIN files, that I generated myself using DGUS_V7641, solved the audio issue on my CR-10s Pro.

I'm currently using FW 7.4.7 with the V7 DWIN files, right now its my best option. Newer options such as setting the fan speed are missing from this version of DWIN so I don't have access to that. This is why I would like to get the V8 DGUS project.

mattcontrols, 2023-08-12:
Im also interested in this!

mattcontrols, 2023-08-12:
> The V7 version was available previously and I edited that a bit (adjust baby stepping values). Miraculously the V7 (FW 7.4.6) DWIN files, that I generated myself using DGUS_V7641, solved the audio issue on my CR-10s Pro.
> 

Can I ask what you mean by adjust the  baby stepping values? I'm running the same machine but I didn't edit anything and am always looking to make it better

> Miraculously the V7 (FW 7.4.6) DWIN files, that I generated myself using DGUS_V7641, solved the audio issue on my CR-10s Pro.
> 
What audio issue? Mine says 1 when it's muted

Ericvf, 2024-06-01:
> What audio issue? Mine says 1 when it's muted

For me the sounds are getting mixed. When I press something instead of a beep I get an Applause. xD

@InsanityAutomation
**Please add `DWprj.hmi` for the V8 files. It is really appreciated!**