InsanityAutomation/Marlin #304: Compiling and Leveling problems with DW7.4.7 ans Ender3 V2-BLT-V422

rstalde, 2023-05-31:
I have compiled the DW7.4.7 version for an Ender 3 V2 BLT with a V4.2.2. Board
Thereby 2 problems occurred:
1. compiler error in line 691, port assigned twice.
I could solve the problem myself: the port LCD_SERIAL_PORT must be 3 (by the way, the same error is also present in DW7.4.6 ;))

2. after homing, the hotend is > 10 mm above the print bed, but 5 mm are displayed.
It is not possible to set the hotend to the required height for printing (setting range not sufficient / assumed height wrong).

I have attached the used configuration.h as zip.

[Configuration.zip](https://github.com/InsanityAutomation/Marlin/files/11618369/Configuration.zip)

dhysuiej, 2023-10-25:
Hi there. I hope you don't mind if I ask an unrelated questions. I have an Ender 3 V2 with 4.2.2 motherboard, CR Touch, BTT smart runout sensor, and a touch screen upgrade kit I [bought from Amazon](https://www.amazon.com/gp/product/B09PL33L5Y). It looks like the touch screen only works with the Creality firmware because Marlin doesn't have support for those screens. I'm not sure what DW7.4.7 is for, but there is [this video](https://www.youtube.com/watch?v=5b7nODzSY5U) of a guy that was able to install InsanityAutomation's firmware and make it work on an Ender 3 Max with the same motherboard and screen I have, which give some hopes. Does anyone know whether there is a firmware from InsanityAutomation that supports all those features above?

rstalde, 2023-10-25:
In response to your question, I looked at the current version 7.4.8 (Bleeding).
There, a touchscreen is specified as a special configuration.
However, you would need to have some configuration file customization skills to use these configurations.
Since I myself have no interest in using touchscreens on the printers, I won't bother with this configuration any further, though.
I am currently using the current version of the original Marlin.org software in customization for various Ender 3 / 5 printer configurations. (An extremely unpleasant and widespread bug has been fixed there only a few days ago, hence this step. ;))

dhysuiej, 2023-10-25:
@rstalde thanks a lot.